flynneva/bno055 #5: BNO055 somtimes sends BUS_OVER_RUN_ERROR (0x07) over UART

whatis777, 2021-01-17:
BNO055 sends back BUS_OVER_RUN_ERROR (0x07) over UART.

This seems to be a known issue: https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BNO55-0x07-error-UART/td-p/14765

flynneva, 2021-01-18:
hi @whatis777, good find.

from the link you shared it looks like the recommended course of action is to just resend the read command if connected via UART. within our code I think we could do a similar filter if connected via UART, ignore this 0x07 error and resend a read request.

what do you think?

whatis777, 2021-01-19:
I think we could even stick to the regular data request cycle: as there is no realtime guarantee (due to the system design and underlying OS), and as the data query frequency is quite high it shouldn't be a problem if one query cycle does not produce IMU data... the next regular cycle will provide data again a few milliseconds later

flynneva, 2021-01-28:
@whatis777 finally getting some more bno055's in the mail today, so hopefully should be able to test this some more 😄

closed 2021-02-03 by flynneva