flynneva/bno055 #17: [feature] add full I2C support

flynneva, 2021-02-09:
add full support to the i2c class. @whatis777's contributions have set us up nicely for this...just need to implement now I think.

looks like someone wrote a cpp ros1 package that already does this...should be able to learn from a lot of the logic:

https://discourse.ros.org/t/new-bno055-i2c-driver-for-ros/7940

tcdtech, 2022-06-13:
Howdy.

I've got _an_ I2C implementation working with this library just using smbus on a raspberry pi.

Haven't actually tested it in ROS navigation yet (that's my next job) but the data being published to bno055/imu looks right so I'm 99% sure it's working fine.

Once it's fully tested and cleaned up I can do a pull request but wanted to ask how generic we'd ideally like the I2C implementation to be.

What platforms are being targetted? Linux platforms with direct /dev/i2c-n interfaces will all work with the smbus implementation (RPi, Nvidia Jetson etc). 

I also have some experience with the MCP2221 and FT232n series USB->I2C devices on windows but they require their own python libraries to be included separately. Is it worth writing interfaces for those in as well or is just linux smbus enough do you think?

flynneva, 2022-11-13:
hey @tcdtech, apologies for not replying to your questions here.

> What platforms are being targetted? 

@tcdtech for the most part I am just targeting typical x86 linux (ubuntu) platforms. I personally havent tried this package at all on aarch64 platforms at all but would be happy to accept a PR to officially add support there. As for other embedded boards / platforms I think we'd have to leave that up to individuals to add support for as we cant support everything 😅 

> Is it worth writing interfaces for those in as well or is just linux smbus enough do you think?

@tcdtech for a project like this that is open-source and maintained purely by the community / volunteers I think down scoping is the right option here.  Obviously if some strong-willed person wants to try and add support to some unique platform there is nothing stopping them 😄

asymingt, 2022-12-06:
See: https://github.com/flynneva/bno055/pull/52

Somebody will need to test if I broke UART though :)

flynneva, 2023-01-14:
Closed by #52

closed 2023-01-14 by flynneva