flynneva/bno055 #18: orientation quaternion is not normalized

cocodmdr, 2021-03-01:
Hello,

I see that the orientation quaternion output of imu is not normalized : it looks like there is a factor 2**14 applied.

output of `ros2 topic echo /bn055/imu` :
`
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: bno055
orientation:
  x: 535.0
  y: -275.0
  z: 4725.0
  w: -15676.0
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.07777777777777778
  y: -0.027777777777777776
  z: -0.2688888888888889
angular_velocity_covariance:
- -1.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
linear_acceleration:
  x: 0.061
  y: -0.059
  z: -0.03
linear_acceleration_covariance:
- -1.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
---
` 

Is that normal? 
Doesn't it cause problem when using with other ROS2 nodes ?

Thank you !

flynneva, 2021-03-03:
first off, great find @cocodmdr!

to be completely honest i dont know much about quaternions but looking at the docs it seems you are right! quaternions should be normalized before output.

ill see what I can do today to make a PR to fix this. ill tag you in the PR just in case you'd like to test it as well.

flynneva, 2021-03-03:
@cocodmdr opened a PR to hopefully address this issue. ill wait a day or so before merging and then go ahead and merge/release

cocodmdr, 2021-03-04:
I just tested it and it looks good

closed 2021-03-04 by cocodmdr