flynneva/bno055 #22: calibration file

cocodmdr, 2021-03-04:
Is it possible to retrieve a calibration file and upload it to the IMU every time we start the device again ?

See this ros driver [section 6](http://wiki.ros.org/ros_imu_bno055) 
> This will generate a calibration file that will be used by the driver.

flynneva, 2021-03-04:
right now i believe you can configure a few of the [calibration parameters within the parameters yaml file](https://github.com/flynneva/bno055/blob/398e99b455d91f0ee5837f9ccdde9a9b38ac0b8b/bno055/params/bno055_params.yaml#L13).

the numbers are in hex unfortunately. maybe we could switch it so the input is in decimal to help the user.

cocodmdr, 2021-03-04:
Ok, thank you, I will have a look.
But how do you get these parameters once you have done the calibration like [here](https://www.youtube.com/watch?v=Bw0WuAyGsnY)?

flynneva, 2021-03-04:
good point. ill look into it and see if its something we can add, a "calibration mode" or something to auto-generate them.

flynneva, 2021-03-04:
in the mean time, check out adafruit's tutorial on how to calibrate the IMU: https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/device-calibration

cocodmdr, 2021-03-05:
Looking in adafruit [library](https://github.com/adafruit/Adafruit_BNO055/blob/1c06d4fa6584e7cc688e3f2a496b1385769bc13a/examples/restore_offsets/restore_offsets.ino#L123) it looks like there are two more parameters for the calibration that are not used in this driver : Accel radius and Mag radius

flynneva, 2021-03-05:
good find, I'll see if I can add those in the pr that saves the calib data to a file

guru-florida, 2021-12-06:
On successful calibration you could read calib data from the device and write the values back to the parameters. We can then read them using "ros2 param dump <nodename>"

flynneva, 2022-11-13:
Closing due to inactivity

closed 2022-11-13 by flynneva