flynneva/bno055 #33: No covariances in output message

MCFurry, 2021-11-09:
Thanks for your great contribution with an updated ROS2 driver.
Unfortunately it seems the output topic does not contain valid covariance values?
This makes merging this imu's data in a robot_localization node impossible...

Would it be possible to add this?

flynneva, 2021-11-11:
@MCFurry thanks for the issue! from some quick googling it looks like an older ROS 1 package someone added some covariance values [they had calculated](https://github.com/Octanis1/bosch_imu_driver/commit/d1132e27ecff46a63c128f7ecacc245c98b2811a). Now I'm not sure how accurate they are...but maybe adding these in there would be better than nothing?

Interested to hear your thoughts!

flynneva, 2021-11-11:
I think ideally these values should not be hard coded but calculated / calibrated on a unit by unit basis

flynneva, 2021-11-11:
found [a similar question on ROS answers](https://answers.ros.org/question/277670/calculating-imu-variance-covariance-matrix/) with an interesting answer that [points to a good article on how to calculate covariance values](https://www.johndcook.com/blog/standard_deviation/) from a streaming data source.

Following [this article](https://www.johndcook.com/blog/standard_deviation/) we should be able to add some code to this repo and calculate covariance values accurately rather than hard coding them.

MCFurry, 2021-11-11:
Indeed, something is definitely better than putting zeroes in the covariance matrix.

The article looks interesting! 
But what you also see a lot is that the calibrated standard deviations, either from datasheet or from calibration are taken as a constant covariance. These covariances are than exposed via parameter such that the user can decide to override them. For example after doing a fresh calibration while mounted.
Example: https://github.com/bluespace-ai/bluespace_ai_xsens_ros_mti_driver/blob/b83d0ec260b870bd4efb50682c4c895a099f6dd6/param/xsens_mti_node.yaml#L49

PR #36 (2021-11-16): [33] add default covariance values, make them configurable

- closes #33
- closes #37  

Description:

- adds default covariance values
- add ability to configure them if desired
- add logic to actually set calibration offsets
closed 2021-11-16 by flynneva