flynneva/bno055 #38: Error orientation_covariance must be float array [9]

3mrrrx, 2021-11-25:
running the sensor on ros2-foxy i got this error for all covariance fields: 

 Receiving sensor data failed with AssertionError:"The 'orientation_covariance' field must be a set or sequence with length 9 and each value of type 'float'"

`        # TODO: make this an option to publish?
        imu_raw_msg.orientation_covariance = [
            self.param.variance_orientation, 0 , 0,
            0, self.param.variance_orientation, 0,
            0, 0, self.param.variance_orientation
        ]`

this must be corrected to for all variance msgs:

`
        xx = self.param.variance_orientation.value[0] # struct.unpack('!f', bytes.fromhex(str(self.param.variance_orientation)))[0]
        yy = self.param.variance_orientation.value[1] # struct.unpack('!f', bytes.fromhex(str(self.param.variance_orientation)))[0]
        zz = self.param.variance_orientation.value[0] # struct.unpack('!f', bytes.fromhex(str(self.param.variance_orientation)))[0]
        imu_raw_msg.orientation_covariance = [
            xx, 0.0 , 0.0,
            0.0, yy, 0.0,
            0.0, 0.0, zz
        ]
`

flynneva, 2021-11-25:
@3mrrrx good find! it sounds like you've already implemented a fix for this bug - would you be able to make a PR so others could also use the fix? if not, no worries I can implement it also from your recommendations here.

thanks again for using this package! very cool seeing others finding it useful

3mrrrx, 2021-11-26:
Hey.. Yes looking forward to use this.

One question about testing.

Did you ever try this library with the context or a robot?

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

billymccafferty, 2021-11-30:
I'm running into the same issue with the latest code. @3mrrrx , it looks like the code you pasted had some comments and lacks line-wrapping, so my copy/paste I believe is not correct. Would you be able to copy the code snippet with line breaks or submit a PR? Thanks so much!!

flynneva, 2021-12-06:
@3mrrrx @billymccafferty could either of you try out [this branch](https://github.com/flynneva/bno055/pull/39) and see if it solves this issue?

billymccafferty, 2021-12-07:
Thanks for the PR to check @flynneva ... the attempt to fix this is much appreciated! :) After pulling the test branch, I'm still receiving the following:

[WARN] [1638838100.427692555] [bno055]: Receiving sensor data failed with AssertionError:"The 'orientation_covariance' field must be a set or sequence with length 9 and each value of type 'float'" 

Looking at the branch, it looks like you corrected imu_raw_msg.linear_acceleration and imu_raw_msg.angular_velocity, and a couple others, but the warning seems to be focused on an issue with the setting of orientation_covariance.

flynneva, 2021-12-07:
thanks to the great work of @billymccafferty I think we have a fix merged into `develop` for this issue.

I'll go ahead and bring it over to `main`, cut a new release and `bloom` it so the binaries have this fix as well

closed 2021-12-07 by flynneva

billymccafferty, 2021-12-07:
Thanks! Great node you've put together!

[image: Welcome to Codai, Inc!]
*Billy McCafferty*
Software shokunin
t: (303) 589-2806 | e: ***@***.***
http://www.codaiinc.com/

PR #40 (2021-12-07): [38] covariance must be float array
Fully tested resolution of remaining covariance params

closes #38