flynneva/bno055 #46: trouble connecting to the bno055

torn8to, 2022-08-08:
plugged into a bno055 hitlego cp2104 into USB to uart converter into the USB port into a USB device.  

/dev/ttyUSB0 does exist but i needed to delete brltty
[image]
[image]

I get the error message on both that the device is not connected even when 5volts are going through the device it is also cool at this time with about 1.5v being dropped after passing through the device.

flynneva, 2022-08-10:
hey @torn8to! sorry you are having trouble (so did I when I first tried connecting to a bno055).

Have you double checked that you have selected UART mode for the bno055? See  [this section](https://github.com/flynneva/bno055#selecting-connection-type) of the README.

Also, just to be sure double check [the pins are connected properly](https://github.com/flynneva/bno055#selecting-connection-type) and are making a good connection.

As a final recommendation: do you have a second bno055 to test? I have fried one before during initial setup (connecting wrong pins to each other) and it never turned on after that.  You may have a dead bno055 in that case and need to get a replacement.

flynneva, 2022-11-13:
Closing due to in-activity. Feel free to reopen if this is still an issue for you

closed 2022-11-13 by flynneva