flynneva/bno055 #55: Testing I2C connection on Raspberry Pi 3B

slgrobotics, 2023-01-14:
Thanks for implementing I2C!
While testing it - here are the issues.

1. Build and run sequence - RPi 3b, Ubuntu 22.04 Server 64 bit, ROS2 Humble installed via apt/binaries
         "i2cdetect -y 1" shows address 0x28

mkdir -p ~/bno055_ws/src
cd ~/bno055_ws/src/
git clone https://github.com/flynneva/bno055.git
cd ..
colcon build
vi ~/bno055_ws/src/bno055/bno055/params/bno055_params_i2c.yaml   - change i2c_bus to 1. Use i2cdetect -y 1
sudo pip3 install smbus
 
source ~/bno055_ws/install/setup.bash
ros2 run bno055 bno055  --ros-args --params-file ~/bno055_ws/src/bno055/bno055/params/bno055_params_i2c.yaml

Issues:
1. Immediately the node crashes - not able to find parameters "i2c_bus" and "i2c_addr":

[INFO] [1673704231.527974952] [bno055]: Initializing parameters
[INFO] [1673704231.595604396] [bno055]: Parameters set to:
[INFO] [1673704231.600864898] [bno055]:         ros_topic_prefix:       "bno055/"
[INFO] [1673704231.605883787] [bno055]:         connection_type:        "i2c"
[INFO] [1673704231.611069341] [bno055]:         frame_id:               "bno055"
[INFO] [1673704231.615995938] [bno055]:         data_query_frequency:   "100"
[INFO] [1673704231.620986284] [bno055]:         calib_status_frequency: "0.1"
[INFO] [1673704231.625980225] [bno055]:         operation_mode:         "12"
[INFO] [1673704231.630957863] [bno055]:         placement_axis_remap:   "P2"
[INFO] [1673704231.635912116] [bno055]:         acc_factor:             "100.0"
[INFO] [1673704231.641059389] [bno055]:         mag_factor:             "16000000.0"
[INFO] [1673704231.646096194] [bno055]:         gyr_factor:             "900.0"
[INFO] [1673704231.651560445] [bno055]:         set_offsets:            "False"
[INFO] [1673704231.656551052] [bno055]:         offset_acc:             "[65516, 165, 65512]"
[INFO] [1673704231.661637753] [bno055]:         radius_acc:             "1000"
[INFO] [1673704231.666519402] [bno055]:         offset_mag:             "[65460, 65182, 637]"
[INFO] [1673704231.671388499] [bno055]:         radius_mag:             "0"
[INFO] [1673704231.676251398] [bno055]:         offset_gyr:             "[2, 65535, 65535]"
[INFO] [1673704231.681657785] [bno055]:         variance_acc:           "[0.017, 0.017, 0.017]"
[INFO] [1673704231.686587142] [bno055]:         variance_angular_vel:   "[0.04, 0.04, 0.04]"
[INFO] [1673704231.691538791] [bno055]:         variance_orientation:   "[0.0159, 0.0159, 0.0159]"
[INFO] [1673704231.696669450] [bno055]:         variance_mag:           "[0.0, 0.0, 0.0]"
[INFO] [1673704231.702505471] [bno055]: ROS node shutdown
[INFO] [1673704231.707485557] [bno055]: No timers to shutdown
Traceback (most recent call last):
  File "/home/ros/bno055_ws/install/bno055/lib/bno055/bno055", line 33, in <module>
    sys.exit(load_entry_point('bno055===0.4.0.', 'console_scripts', 'bno055')())
  File "/home/ros/bno055_ws/install/bno055/lib/python3.10/site-packages/bno055/bno055.py", line 94, in main
    node.setup()
  File "/home/ros/bno055_ws/install/bno055/lib/python3.10/site-packages/bno055/bno055.py", line 70, in setup
    self.param.i2c_bus.value,
AttributeError: 'NodeParameters' object has no attribute 'i2c_bus'
[ros2run]: Process exited with failure 1

========= OK, at this point I just edited "src/bno055/bno055/bno055.py" to see if it will work at all:
        elif self.param.connection_type.value == I2C.CONNECTIONTYPE_I2C:
            connector = I2C(self,
                            1,
                            0x28)
        else:
======================================

2. It worked, and produces valid data in topics (or, at least in rqt there are values coming at 70Hz and calibration status makes sense). But there's a division by zero somewhere:

ros@turtle:~/bno055_ws$ ros2 run bno055 bno055  --ros-args --params-file ~/bno055_ws/src/bno055/bno055/params/bno055_params_i2c.yaml
[INFO] [1673705369.207049590] [bno055]: Initializing parameters
[INFO] [1673705369.276023821] [bno055]: Parameters set to:
[INFO] [1673705369.280943023] [bno055]:         ros_topic_prefix:       "bno055/"
[INFO] [1673705369.286472482] [bno055]:         connection_type:        "i2c"
[INFO] [1673705369.292932198] [bno055]:         frame_id:               "bno055"
[INFO] [1673705369.302135290] [bno055]:         data_query_frequency:   "100"
[INFO] [1673705369.308729381] [bno055]:         calib_status_frequency: "0.1"
[INFO] [1673705369.314610350] [bno055]:         operation_mode:         "12"
[INFO] [1673705369.320466214] [bno055]:         placement_axis_remap:   "P2"
[INFO] [1673705369.328608372] [bno055]:         acc_factor:             "100.0"
[INFO] [1673705369.336255116] [bno055]:         mag_factor:             "16000000.0"
[INFO] [1673705369.341144056] [bno055]:         gyr_factor:             "900.0"
[INFO] [1673705369.346109664] [bno055]:         set_offsets:            "False"
[INFO] [1673705369.351926934] [bno055]:         offset_acc:             "[65516, 165, 65512]"
[INFO] [1673705369.358342120] [bno055]:         radius_acc:             "1000"
[INFO] [1673705369.363342050] [bno055]:         offset_mag:             "[65460, 65182, 637]"
[INFO] [1673705369.368165418] [bno055]:         radius_mag:             "0"
[INFO] [1673705369.373985866] [bno055]:         offset_gyr:             "[2, 65535, 65535]"
[INFO] [1673705369.381897139] [bno055]:         variance_acc:           "[0.017, 0.017, 0.017]"
[INFO] [1673705369.389718570] [bno055]:         variance_angular_vel:   "[0.04, 0.04, 0.04]"
[INFO] [1673705369.396823284] [bno055]:         variance_orientation:   "[0.0159, 0.0159, 0.0159]"
[INFO] [1673705369.403196854] [bno055]:         variance_mag:           "[0.0, 0.0, 0.0]"
[INFO] [1673705369.467421364] [bno055]: Configuring device...
[INFO] [1673705369.475430866] [bno055]: Current sensor offsets:
[INFO] [1673705369.485973172] [bno055]:         Accel offsets (x y z): 136 40 65371
[INFO] [1673705369.493317729] [bno055]:         Accel radius: 33768
[INFO] [1673705369.499668956] [bno055]:         Mag offsets (x y z): 128 0 0
[INFO] [1673705369.504602168] [bno055]:         Mag radius: 128
[INFO] [1673705369.509957669] [bno055]:         Gyro offsets (x y z): 129 1 0
[INFO] [1673705369.514781766] [bno055]: Setting device_mode to 12
[INFO] [1673705369.520585339] [bno055]: Bosch BNO055 IMU configuration complete.
[WARN] [1673705369.550428149] [bno055]: Receiving sensor data failed with ZeroDivisionError:"float division by zero"
[WARN] [1673705369.571708020] [bno055]: Receiving sensor data failed with ZeroDivisionError:"float division by zero"
[WARN] [1673705369.593917524] [bno055]: Receiving sensor data failed with ZeroDivisionError:"float division by zero"

Hope this helps.

slgrobotics, 2023-01-14:
Related to https://github.com/flynneva/bno055/pull/52

flynneva, 2023-01-14:
@slgrobotics awesome! thanks for reporting this, I'll look into this right now

flynneva, 2023-01-14:
@slgrobotics this error comes from the `SensorService.py` file, specifically the `get_sensor_data` function. We should probably add some checks to ensure we dont divide by zero in order to avoid this error.

I'll make a PR and tag you in it if thats ok?

flynneva, 2023-01-14:
@slgrobotics could you test out #56 to see if that fixes these two errors?

PR #56 (2023-01-15): Fix i2c issues found (param and division by zero)
Closes #55
closed 2023-01-15 by flynneva