flynneva/bno055 #63: Communication error using UART to USB

Danilrivero, 2023-04-20:
Hello,

I have bumped into the following issue when trying to connect to the BNO055. I made sure it properly works beforehand by using a small Arduino snippet and it does. 

I get the following message when running the node configured with the _yaml_ file.

[image]

CP2104 and connections work properly and the tty/USB0 port is properly attached:

[image]

I am using **Ubuntu 22.04 with ROS2 Humble** and all the libraries required to run the driver are installed.

The only parameter changed is the baudrate due to my requirements:

[image]

**Has anyone bumped into this issue before? Any advice on how to proceed?** The response header **0xff** does not give me enough information to deal with it.

Danilrivero, 2023-04-21:
To add more information:

- If I use a **baudrate** of 115200 I get the following error:

[image]

- Drivers for the USB to UART adapter seem to be installed properly:

[image]

- Tried giving rw permissions to tty/USB0 manually as well:

[image]

I can send more information if required.

flynneva, 2023-04-24:
hey @Danilrivero! I haven't actually had a bno055 to test this driver in awhile - but from what I can tell the last change to the UART side of things for this driver was [this commit](https://github.com/flynneva/bno055/commit/01abe0371017f28fb2dabb0b6bc6ddd30270dac2). Could you maybe try to revert that change and see if UART works?

I really need to get around to adding some integration tests to this repo...but havent had much time unfortunately.

Danilrivero, 2023-04-24:
Hello @flynneva that seems to be the issue. Thanks for your comment!

Closing the issue due to the fix being found.

flynneva, 2023-04-25:
@Danilrivero ok awesome. I'll go ahead and remove this change from main then, thanks for confirming that reverting that line fixed it!

flynneva, 2023-04-25:
@emilnovak see the thread above. Since you authored the problem commit here, could you give some background on why you made that change / what it fixed for you?

Maybe we are just missing an if / else somewhere

emilnovak, 2023-11-17:
Sorry for the late reply @flynneva, I completely missed this thread. For future reference the reasoning can be found here: [#52 (comment)](https://github.com/flynneva/bno055/pull/52#discussion_r1081186341). I may be able to look further into this problem, since I'm actively using some BNO sensors with this change via serial.

flynneva, 2023-11-20:
@emilnovak no worries! I know how it goes - I haven't touched this repo in a long time either. I might have some time this week to put a bit of work in here as well...let's see 👍

shrujanus, 2025-02-07:
> Hello [@flynneva](https://github.com/flynneva) that seems to be the issue. Thanks for your comment!
> 
> Closing the issue due to the fix being found.

Hey @Danilrivero I am having the same issue can you share how you solved it?