flynneva/bno055 #68: Interest in tagging a new release?

traversaro, 2024-02-10:
Thanks a lot @flynneva for working on this repo! As it passed some time since the last release (0.4.1) and some fixes (such as https://github.com/flynneva/bno055/pull/59) are not yet released, are you interested in making a new release, also in view of ROS 2 Jazzy? 

Thanks a lot in advance!

flynneva, 2024-02-12:
@traversaro sure! I'll cut a new one this week 👍

traversaro, 2024-02-12:
> @traversaro sure! I'll cut a new one this week 👍

Thanks!

flynneva, 2024-02-17:
@traversaro I've tagged and opened up rosdistro PRs for all active ROS 2 distros (here is the rolling PR: https://github.com/ros/rosdistro/pull/39928).

Now we wait for them to get reveiwed / merged and then for a patch release for each distro.

Going to close this issue as complete 👍🏼

closed 2024-02-17 by flynneva

traversaro, 2024-02-17:
Thanks a lot!