flynneva/bno055 #71: Calibrated Data

sagar0609nagdev, 2024-04-11:
Hello Evan, I have successfully cloned this repository and data is being published. I want to use this in LIO-SAM algorithm but before that I need to calibrate it. 

So I have calibrated data separately, I want to ask how can I use this data in your code or were can I add that so that whenever I start your code I get calibrated data.
Or can you suggest me any other way for getting calibrated data directly from your algorithm.
Thank you.

flynneva, 2024-04-16:
@sagar0609nagdev calibration is one area where I haven't yet really dove into yet but would be interested in.

Doing a quick search it looks like bno055 devices should calibrate themselves I think, and just need to be moved in a specific way while on to calibrate:

 https://www.mathworks.com/help/matlab/supportpkg/calibrate-sensors.html

I'm traveling today and at the airport, but let me know if you figure something out with the link above. We can add this to the readme for this repo once we learn more, so others can follow the steps too.

sagar0609nagdev, 2024-04-25:
Hello, sorry for delay. 
With this the imu shows this on terminal,  for e.g. data: '{"sys": 3, "gyro": 3, "accel": 3, "mag": 3}' but after some time it changes 
data: '{"sys": 1, "gyro": 3, "accel": 3, "mag": 3}. So I don't know if its fully calibrated or not.

Also everytime I start IMU this thing pops up on terminal:
[bno055-1] [WARN] [1714049071.538192834] [bno055]: Receiving calibration status failed with BusOverRunException:"Data fusion not ready, resend read request".