flynneva/bno055 #73: Refresh Rate only 48Hz not 100Hz

PhilShaw852, 2024-06-20:
Raspberry Pi Zero 2 W and Pi 3B
Ubuntu 22.04
Ros Humble

Using I2C the topic hz only reaches 48Hz, when set to 100. Can this be improved? With a Pi 5 it reaches 100hz.

PhilShaw852, 2024-06-21:
[image]
It appears that the issue is that the node causes the core to hit 100% usage. If I reduce the refresh rate the CPU usage drops.

Should it be this CPU hungry?

flynneva, 2024-06-22:
@PhilShaw852 seems to be an issue with clock stretching on the pi3 (someone had the same issue as you it looks like back in 2017)

https://forums.raspberrypi.com/viewtopic.php?p=1231112#p1231112

PhilShaw852, 2024-06-22:
Hi, I have tried using the software i2c and have a similar maximum refresh rate when viewing via ros2 topic hz /bno055/imu.

With the software i2c it looks like the core is the limiting factor as the core running the node shows 100% usage.

flynneva, 2024-06-25:
@PhilShaw852 oof ok. There might be ways to optimize the code a bit since its so old now 😅 if I can find some time I'll try and see if I can find any places where we can improve