flynneva/bno055 #74: UART Communication error: Wrong READ-request response header 0xdb

PhilShaw852, 2024-06-20:
Pi 3B and Zero 2 W
Ubuntu 22.04
Ros Humble

`ros2 run bno055 bno055 --ros-args --params-file ./ros2_ws/src/bno055/bno055/params/bno055_params.yaml
[INFO] [1718869957.803336347] [bno055]: Initializing parameters
[INFO] [1718869957.893408861] [bno055]: Parameters set to:
[INFO] [1718869957.902802641] [bno055]: 	ros_topic_prefix:	"bno055/"
[INFO] [1718869957.911990745] [bno055]: 	connection_type:	"uart"
[INFO] [1718869957.921358327] [bno055]: 	uart_port:		"/dev/ttyS0"
[INFO] [1718869957.930839241] [bno055]: 	uart_baudrate:		"115200"
[INFO] [1718869957.940737392] [bno055]: 	uart_timeout:		"0.1"
[INFO] [1718869957.950334712] [bno055]: 	frame_id:		"bno055"
[INFO] [1718869957.959898542] [bno055]: 	data_query_frequency:	"100"
[INFO] [1718869957.969348571] [bno055]: 	calib_status_frequency:	"0.1"
[INFO] [1718869957.978788548] [bno055]: 	operation_mode:		"12"
[INFO] [1718869957.988214047] [bno055]: 	placement_axis_remap:	"P2"
[INFO] [1718869957.998071520] [bno055]: 	acc_factor:		"100.0"
[INFO] [1718869958.007484883] [bno055]: 	mag_factor:		"16000000.0"
[INFO] [1718869958.017044026] [bno055]: 	gyr_factor:		"900.0"
[INFO] [1718869958.026273640] [bno055]: 	grav_factor:		"100.0"
[INFO] [1718869958.035527890] [bno055]: 	set_offsets:		"False"
[INFO] [1718869958.044982086] [bno055]: 	offset_acc:		"[65516, 165, 65512]"
[INFO] [1718869958.054401907] [bno055]: 	radius_acc:		"1000"
[INFO] [1718869958.064082663] [bno055]: 	offset_mag:		"[65460, 65182, 637]"
[INFO] [1718869958.073658004] [bno055]: 	radius_mag:		"0"
[INFO] [1718869958.083114283] [bno055]: 	offset_gyr:		"[2, 65535, 65535]"
[INFO] [1718869958.092448792] [bno055]: 	variance_acc:		"[0.017, 0.017, 0.017]"
[INFO] [1718869958.101737417] [bno055]: 	variance_angular_vel:	"[0.04, 0.04, 0.04]"
[INFO] [1718869958.111175467] [bno055]: 	variance_orientation:	"[0.0159, 0.0159, 0.0159]"
[INFO] [1718869958.120579038] [bno055]: 	variance_mag:		"[0.0, 0.0, 0.0]"
[INFO] [1718869958.131411400] [bno055]: Opening serial port: "/dev/ttyS0"...
[INFO] [1718869958.226212471] [bno055]: Configuring device...
[ERROR] [1718869958.336322639] [bno055]: Communication error: Wrong READ-request response header 0xdb
[ERROR] [1718869958.346387663] [bno055]: Shutting down ROS node...
[INFO] [1718869958.357300962] [bno055]: ROS node shutdown
[INFO] [1718869958.366553909] [bno055]: No timers to shutdown
[ros2run]: Process exited with failure 1`

Any help would be appreciated.

TheHassanShahzad, 2024-07-28:
same issue here. was first on I2c mode then switched to UART by bridiging S1 + to S1 middle pad for my board and still have this issue.

Before it was perission denied for i2c and doing sudo chmod 777 /dev/ttyUSB0 still didnt fix it

duygiang1162, 2025-01-04:
> same issue here. was first on I2c mode then switched to UART by bridiging S1 + to S1 middle pad for my board and still have this issue.
> 
> Before it was perission denied for i2c and doing sudo chmod 777 /dev/ttyUSB0 still didnt fix it

Hei guy, how did you fix this problem. Thank for your advised